adafruit/Adafruit_HX8357_Library #2: Compatibility with Arduino 101

amirault0, 2016-03-03:
[HX8357_Arduino_101_error.txt](https://github.com/adafruit/Adafruit_HX8357_Library/files/155993/HX8357_Arduino_101_error.txt)

I am unable to compile the graphicstest sketch with an Arduino 101. Does this library support it yet?

ladyada, 2016-03-03:
nope!

amirault0, 2016-03-03:
Are there plans to add official compatibility? I've managed to get the sketch to compile and upload, but the LCD does not do anything.

ladyada, 2016-11-16:
if you have an arduino 101 handy you can try the latest code in github, it compiles - *probably* works :)

ladyada, 2017-09-23:
we just refactored to match the ILI9341 so it almost certainly works with 101 but since the 101 is discontinued we aren't going to test - we'll be closing this issue - please reopen if it still doesnt work!

closed 2017-09-23 by ladyada