adafruit/Adafruit_HX8357_Library #4: Compatiblity with Adafruit Feather M0

STrRedWolf, 2016-04-24:
Trying to use the board with the Feather M0 and having these compile issues (using the latest library dated today):

`````
Arduino: 1.6.8 (Windows 10), Board: "Adafruit Feather M0 (Native USB Port)"

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp: In constructor 'Adafruit_HX8357::Adafruit_HX8357(int8_t, int8_t, int8_t, int8_t, int8_t, int8_t)':

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:30:3: error: '_cs' was not declared in this scope

   _cs   = cs;

   ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:31:3: error: '_dc' was not declared in this scope

   _dc   = dc;

   ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:32:3: error: '_mosi' was not declared in this scope

   _mosi  = mosi;

   ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:33:3: error: '_miso' was not declared in this scope

   _miso = miso;

   ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:34:3: error: '_sclk' was not declared in this scope

   _sclk = sclk;

   ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:35:3: error: '_rst' was not declared in this scope

   _rst  = rst;

   ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp: In constructor 'Adafruit_HX8357::Adafruit_HX8357(int8_t, int8_t, int8_t)':

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:43:3: error: '_cs' was not declared in this scope

   _cs   = cs;

   ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:44:3: error: '_dc' was not declared in this scope

   _dc   = dc;

   ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:45:3: error: '_rst' was not declared in this scope

   _rst  = rst;

   ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:47:3: error: '_mosi' was not declared in this scope

   _mosi  = _sclk = 0;

   ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:47:12: error: '_sclk' was not declared in this scope
[... 372 lines cut ...]

    digitalWrite(_dc, LOW);

                 ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:581:17: error: '_sclk' was not declared in this scope

    digitalWrite(_sclk, LOW);

                 ^

C:\Users\tygris\Documents\Arduino\libraries\Adafruit_HX8357\Adafruit_HX8357.cpp:582:17: error: '_cs' was not declared in this scope

    digitalWrite(_cs, LOW);

                 ^

exit status 1
Error compiling for board Adafruit Feather M0 (Native USB Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

`````

ladyada, 2016-04-24:
yep, its not ported to the SAMD21 yet - you do figure out we'll merge. no ETA on us doing it at this time.

STrRedWolf, 2016-04-24:
Do we know what #define's are set for the SAMD21 in gcc?  That'll help me in the long run if I'm forking it.

ladyada, 2016-04-24:
check out
https://github.com/adafruit/Adafruit-SSD1351-library/pull/10/commits/111055c4f6e194020a4313a2be6f5bfc4e2b08b5

and 

https://github.com/adafruit/Adafruit_ILI9341/blob/master/Adafruit_ILI9341.h#L159

for ideas?

STrRedWolf, 2016-04-26:
I have it working with SPI, but it does not work with the SPI based card reader.  I think the M0's SPI SERCOM doesn't like to multiplex with arbitrary pins.

I'm now looking into moving the SPI to exclusive card-only and use the 8-bit mode.  More research!

smartrip, 2016-11-10:
Hi guys, @STrRedWolf can you publish your code for TFT in SPI mode on Feather M0? I'm working with the library and also use SAMD21 for my project. I don't need SD card reader.

Thanks!!!

STrRedWolf, 2016-11-11:
My code is a bit of a mess, alas, and it's been a bit too long.  But @ladyada did point to the right places to adjust in order for it to work.

I also hope @ladyada releases a version of the 320x480 display as a FeatherWing. That'll fit my use case and maybe yours.

smartrip, 2016-11-12:
Thanks @STrRedWolf - I ended up with 2.8" ILI 9341 display that works perfectly well with M0 (Feather/Zero) and the touchscreen that is driven by I2C, also fully supported. Clean config and no need to port the drivers.

ladyada, 2016-11-16:
fixzed!

closed 2016-11-16 by ladyada