adafruit/Adafruit_HX8357_Library #9: NEW HX8357 LIBRARY - COMPLIER ERRORS WITH ESP32 FEATHER

rjp663, 2018-05-09:
After installing the new revision of the HX8357 Library using the Arduino IDE, compiler errors are occurring. Returning to the previous version corrects the error.

techjeffharris, 2018-05-14:
I am also experiencing the same issue with version 1.0.7.  Here is the verbose logging: 

```Arduino: 1.8.5 (Linux), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None"

/home/jeff/Downloads/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/jeff/Downloads/arduino-1.8.5/hardware -hardware /home/jeff/.arduino15/packages -hardware /home/jeff/Arduino/hardware -tools /home/jeff/Downloads/arduino-1.8.5/tools-builder -tools /home/jeff/Downloads/arduino-1.8.5/hardware/tools/avr -tools /home/jeff/.arduino15/packages -built-in-libraries /home/jeff/Downloads/arduino-1.8.5/libraries -libraries /home/jeff/Arduino/libraries -fqbn=espressif:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10805 -build-path /tmp/arduino_build_972634 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /home/jeff/Arduino/tft_feather_esp32_battery_monitor/tft_feather_esp32_battery_monitor.ino
/home/jeff/Downloads/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/jeff/Downloads/arduino-1.8.5/hardware -hardware /home/jeff/.arduino15/packages -hardware /home/jeff/Arduino/hardware -tools /home/jeff/Downloads/arduino-1.8.5/tools-builder -tools /home/jeff/Downloads/arduino-1.8.5/hardware/tools/avr -tools /home/jeff/.arduino15/packages -built-in-libraries /home/jeff/Downloads/arduino-1.8.5/libraries -libraries /home/jeff/Arduino/libraries -fqbn=espressif:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10805 -build-path /tmp/arduino_build_972634 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /home/jeff/Arduino/tft_feather_esp32_battery_monitor/tft_feather_esp32_battery_monitor.ino
Using board 'featheresp32' from platform in folder: /home/jeff/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /home/jeff/Arduino/hardware/espressif/esp32
Detecting libraries used...
"/home/jeff/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/heap" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/spiffs" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/coap" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/console" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/expat" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/json" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/soc" "-I/home/jeff/Arduino/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_FEATHER_ESP32 -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="FEATHER_ESP32" -DARDUINO_VARIANT="feather_esp32"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-I/home/jeff/Arduino/hardware/espressif/esp32/cores/esp [... cut ...]
Using library FS at version 1.0 in folder: /home/jeff/Arduino/hardware/espressif/esp32/libraries/FS 
Using library SD at version 1.0.5 in folder: /home/jeff/Arduino/hardware/espressif/esp32/libraries/SD 
Using library SPI at version 1.0 in folder: /home/jeff/Arduino/hardware/espressif/esp32/libraries/SPI 
Using library WiFi at version 1.0 in folder: /home/jeff/Arduino/hardware/espressif/esp32/libraries/WiFi 
Using library ResponsiveAnalogRead at version 1.1.0 in folder: /home/jeff/Arduino/libraries/ResponsiveAnalogRead 
exit status 1
Error compiling for board Adafruit ESP32 Feather.
```

ladyada, 2018-05-20:
should be good now

closed 2018-05-20 by ladyada

Eagle334, 2019-02-03:
Hello. With the 1.7, i have the same error message. Compil is OK with 1.6.