adafruit/Adafruit_HX8357_Library #10: Issue displaying bmp images with ESP32 Feather with WiFi.begin() function

IoTAll, 2018-05-10:
Hi,
The bmp image using the example from the library gets blurred as soon as the function WiFi.begin() is called.
I am using the Feather ESP32 with the 3.5" tft Feather wing touch screen.
If I don't call this WiFi.begin function, the picture is displayed correctly.

Is it something coming from the library?

Thank you!

techjeffharris, 2018-05-14:
Same Issue here!

IoTAll, 2018-05-16:
Issue was coming from the TFT_CS pin which must not be connected to an ADC2 pin if WiFi is connected. Moving it to pin 21 on the ESP32 solve the issue. 
A note in the Adafruit tutorial for this library used with the ESP32 would be great to have!

ladyada, 2018-05-17:
oh jeez ok that explains it - ok we'll document it in the guide!

ladyada, 2018-05-17:
OK i dont have an HX but i have a ili 2.4" featherwing and WiFi.begin() doesnt make any difference so its *not* the pin

BTLOPER, 2018-06-28:
Greetings My @ladyada,

I've been reading about this issue today trying to work an unrelated problem.  

According to your pinout page for the Huzzah32 feather:
"15 - This is GPIO #15 and also an analog input A8 on ADC #2"

This thread from your forum descibes the "solution:"
[Issue with ESP32 bmpDraw and WiFi.begin() ](https://forums.adafruit.com/viewtopic.php?f=57&t=134733&p=672550&hilit=ESP32+SPI#p672550)

In short though, as suggested there, ADC2 pins cannot be used at the same time as WiFi.  

Hope that helps.

makermelissa, 2019-04-25:
There have been some recent updates to this driver. I just tried WiFi.begin() right after tft.begin() and had no problems loading a bitmap using the ImageReader example for the HX8357 Featherwing. I would suggest using that to load images. If you are still experiencing problems, please feel free to reopen this issue. :)

closed 2019-04-25 by makermelissa