adafruit/Adafruit_HX8357_Library #12: please provide examples for button press (Featherwing 3.5")

dsyleixa, 2018-09-14:
please provide examples for button press (Featherwing 3.5"),
a) landscape
b) portrait

dsyleixa, 2018-09-29:
PS,
as I just noticed the PR about a menu system:
The basic example I actually mean is just about:

create 2 different buttons
detect touch position if touched anywhere
show touch (x,y) position on the screen
change the specific button colors if a button was hit once (e.g., inverse) 
write a notification on the screen if a botton was released (i.e., in a proprietary program used to trigger an arbitrary  function by button press)

dsyleixa, 2018-10-02:
Your reply is a little weird, probably you are missing my point:
if I could write touch screen button examples I hadn't written this request for multiple touch screen buttons examples, finally nothing exists at all!

ladyada, 2018-10-02:
ok we'll leave this issue open, no guarantee on when we'll get to writing the exact example you like - we've got many many libraries to manage :)

dsyleixa, 2018-10-24:
Now will there soon be some basic code examples for programming buttons?

ladyada, 2018-10-24:
no ETA on this, we have plenty of bugs to fix in our libraries so this is not a high priority! please don't bump - it won't get it done any faster :) if you want to hire someone to help with custom code needs, check out our jobs board! https://jobs.adafruit.com/

makermelissa, 2019-04-24:
A button example was merged in today.

closed 2019-04-24 by makermelissa