adafruit/Adafruit_HX8357_Library #13: p.x and p.y ts touch positions not detected any more if loops lasts about 1000ms

dsyleixa, 2018-09-14:
this issue is about Feather Huzzah and TFT Featherwing 3.5"

p.x and p.y ts touch positions not detected any more if loops lasts about 1000ms:
```

void loop() {
  int i;
  char buf[20]="", buf1[20]="";
  TS_Point p;
  p.x=-1; p.y=-1;

  
  // Retrieve a point   
  p = ts.getPoint();
  // Scale from ~0->4000 to tft.width using the calibration #'s
  p.x = map(p.x, TS_MINX, TS_MAXX, tft.width(), 0);
  p.y = map(p.y, TS_MINY, TS_MAXY, 0, tft.height());
  delay(1); // debounce   
 
    Serial.print("X = "); Serial.print(p.x); Serial.print("\tY = "); Serial.print(p.y);  Serial.print("\tPressure = "); Serial.println(p.z);           
    Serial.println();    
    sprintf(buf, "x=%3d", p.x); sprintf(buf1, "y=%3d", p.y);
    tft.setTextColor(RED); 
    tft.fillRect(80,0, 80,60, COLOR_BKGR);  // x1,y1, dx,dy 
    tft.setCursor(85,0);     tft.println(buf);
    tft.setCursor(85,20);    tft.println(buf1);
  
  delay(1000);  // real code: long lasting calculations of <= 1000ms  // <<~~~~~~~~~
}
```

OTOH, if there is no delay at line
delay(1000);  // real code: long lasting calculations of <= 1000ms  // <<~~~~~~~~~
 the code works fine, all p postion are detected correctly.

How can this be fixed?

dsyleixa, 2018-09-16:
perhaps it's not a HX8357 issue but a GFX issue.
If so, can that be moved then or shall I create a new issue below GFX?

ladyada, 2018-09-16:
nope - no ETA when we'll get to looking at this

dsyleixa, 2018-09-16:
pardon, I don't understand?
Is it correct here or shall I open a new issue?

ladyada, 2018-09-16:
its fine here - dont keep opening issues :)

dsyleixa, 2018-09-21:
Update:
I just purchased your Feather M4 as well, and for the M4 everything runs like a charm, with exactly the same program:
All touch positions are detected very quick by the M4.

So it's appearently a  touch issue especially for the Huzzah.

dsyleixa, 2018-09-26:
will there be a bugfix for the touchscreen+Huzzah  issue in the near future?

ladyada, 2018-09-26:
no ETA on it, because it works with the default code

dsyleixa, 2018-09-27:
ok, as you wouldn't fix it I dropped the Huzzah thing then, so I have issued a return case to the seller.

makermelissa, 2019-04-25:
There have been recent updates to the library. I was able to successfully run your code on a Feather Huzzah with the delay (after updating the colors to the new values) and it ran just fine. If you are still having trouble, please feel free to reopen this issue. :)

closed 2019-04-25 by makermelissa