adafruit/Adafruit_HX8357_Library #14: how to check for attached boards by "board type names"?

dsyleixa, 2018-09-23:
in your example you check for attached boards via board names, e.g.

```
#ifdef ESP8266
   #define STMPE_CS 16
   #define TFT_CS   0
   #define TFT_DC   15
   #define SD_CS    2
#endif
#ifdef ESP32
   #define STMPE_CS 32
   #define TFT_CS   15
   #define TFT_DC   33
   #define SD_CS    14
#endif
#ifdef TEENSYDUINO
   #define TFT_DC   10
   #define TFT_CS   4
   #define STMPE_CS 3
   #define SD_CS    8
#endif
#ifdef ARDUINO_STM32_FEATHER
   #define TFT_DC   PB4
   #define TFT_CS   PA15
   #define STMPE_CS PC7
   #define SD_CS    PC5
#endif
#ifdef ARDUINO_FEATHER52
   #define STMPE_CS 30
   #define TFT_CS   13
   #define TFT_DC   11
   #define SD_CS    27
#endif
#if defined(ARDUINO_MAX32620FTHR) || defined(ARDUINO_MAX32630FTHR)
   #define TFT_DC   P5_4
   #define TFT_CS   P5_3
   #define STMPE_CS P3_3
   #define SD_CS    P3_2
#endif

// Anything else!
#if defined (__AVR_ATmega32U4__) || defined(ARDUINO_SAMD_FEATHER_M0) || defined (__AVR_ATmega328P__) || defined(ARDUINO_SAMD_ZERO) || defined(__SAMD51__) || defined(__SAM3X8E__)
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5
#endif
```
Unfortunately, some of my boards cannot be detected, i.e. my Adafruit Itsybitsy M0 and Arduino Due programming port. For Feather Huzzah (esp8266) and Feather M4 it works well though.
How can I determine the board type name, in case also other ones still not being checked yet?

dsyleixa, 2018-09-23:
Update:
adding
|| defined(ARDUINO_ITSYBITSY_M0)
it now also works for this one.

HTH for you for a lib update!

ladyada, 2018-09-23:
thats the featherwng example, so it isnt designed for itsybitsy!

closed 2018-09-23 by ladyada

dsyleixa, 2018-09-23:
That is a poor argument!
Eventually the itsybitsy can be attached too, by wires!
And finally also the Arduino Due __SAM3X8E__  can be #defined, and it's also no Feather!

dsyleixa, 2018-09-23:
different aproach:
#elif defined()
plus
#else
for defaults:

```
   #ifdef ESP8266
   #define STMPE_CS 16
   #define TFT_CS   0
   #define TFT_DC   15
   #define SD_CS    2
 
#elif defined ESP32
   #define STMPE_CS 32
   #define TFT_CS   15
   #define TFT_DC   33
   #define SD_CS    14
 
#elif defined TEENSYDUINO
   #define TFT_DC   10
   #define TFT_CS   4
   #define STMPE_CS 3
   #define SD_CS    8
  
#elif defined ARDUINO_STM32_FEATHER
   #define TFT_DC   PB4
   #define TFT_CS   PA15
   #define STMPE_CS PC7
   #define SD_CS    PC5

#elif defined ARDUINO_FEATHER52
   #define STMPE_CS 30
   #define TFT_CS   13
   #define TFT_DC   11
   #define SD_CS    27
 
#elif  defined(ARDUINO_MAX32620FTHR) || defined(ARDUINO_MAX32630FTHR)
   #define TFT_DC   P5_4
   #define TFT_CS   P5_3
   #define STMPE_CS P3_3
   #define SD_CS    P3_2

// Anything else!
#elif  defined (__AVR_ATmega32U4__) || defined(ARDUINO_SAMD_FEATHER_M0) || defined (__AVR_ATmega328P__) || defined(ARDUINO_SAMD_ZERO) || defined(__SAMD51__)   
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5

 // default
#else
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5
#endif

```

ladyada, 2018-09-23:
hiya you can submit a PR if ya like and we’ll take a look next time we do work on this li

dsyleixa, 2018-09-24:
hi,
I have no idea how a PR works, I don't work or publish on github at all, it's far too comlicated.

The former problem was that in your libs you have multiple #ifdef...#endif sections which made it hard to create arbitrary pin definitions without knowing the actual board type names.
Now by my mofification it is possible to add an arbitrary `"#else (default) section"` for already not explicitely mentioned  and 3rd party boards.
Feel free to utilize it for your next lib update!