adafruit/Adafruit_HX8357_Library #16: ESP8266 Compilation issue

ronguest, 2019-03-10:
I am seeing a compile failure specific to the ESP8266 when using the included example graphicstest_featherwing. Reproduced by installing a new Arduino IDE 1.8.8, downloading fresh copies of the GFX and HX8357 libraries, setting the board type to Feather M0 - compiles without errors. Change the board type to Feather HUZZAH 8266 and the compile will fail with the below error. In looking at the GfX repo I see some changes were made last week specifically for the 8266. I looked at the copy of the GfX library source and confirmed the changes are in the copy on my machine. I don’t understand the libraries enough to know if the problem lies in the GfX library itself or HX8357. But since the error is reported on the HX8357 at compilation I’m filing it here.

Here are the compile errors:

> Arduino: 1.8.8 (Mac OS X), Board: "Adafruit Feather HUZZAH ESP8266, 80 MHz, Flash, Disabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Users/ronaldg/Documents/Arduino/libraries/Adafruit_HX8357_Library/Adafruit_HX8357.cpp: In constructor 'Adafruit_HX8357::Adafruit_HX8357(int8_t, int8_t, int8_t, uint8_t, SPIClass*)':
/Users/ronaldg/Documents/Arduino/libraries/Adafruit_HX8357_Library/Adafruit_HX8357.cpp:116:56: error: invalid conversion from 'SPIClass*' to 'int8_t {aka signed char}' [-fpermissive]
   HX8357_TFTHEIGHT, spi, cs, dc, rst), displayType(type) {
                                                        ^
In file included from /Users/ronaldg/Documents/Arduino/libraries/Adafruit_HX8357_Library/Adafruit_HX8357.h:34:0,
                 from /Users/ronaldg/Documents/Arduino/libraries/Adafruit_HX8357_Library/Adafruit_HX8357.cpp:42:
/Users/ronaldg/Documents/Arduino/libraries/Adafruit_GFX_Library/Adafruit_SPITFT.h:121:5: error:   initializing argument 3 of 'Adafruit_SPITFT::Adafruit_SPITFT(uint16_t, uint16_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t)' [-fpermissive]
     Adafruit_SPITFT(uint16_t w, uint16_t h,
     ^
exit status 1
Error compiling for board Adafruit Feather HUZZAH ESP8266.

- Arduino board: Feather Huzzah 8266
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.8
- GfX v1.4.2
- HX8357 v1.1.0
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): See above

ladyada, 2019-03-10:
@PaintYourDragon

caternuson, 2019-03-21:
Same here:
https://forums.adafruit.com/viewtopic.php?f=47&t=148891&start=15#p737251

I was able to recreate. There's a `#if` that removes the constructor for ESP8266.
https://github.com/adafruit/Adafruit-GFX-Library/blob/5f38af166fdb1faf4cea2ffd67857d8b09cd66f5/Adafruit_SPITFT.cpp#L240-L246

PaintYourDragon, 2019-03-22:
Fixed in v1.1.1. Thanks.

closed 2019-03-22 by PaintYourDragon