adafruit/Adafruit_HX8357_Library #17: Flipping the 3.5 FeatherWing Enable Switch off and on causes reversing

makermelissa, 2019-04-20:
I'm not exactly sure why this is happening, but with the 3.5" TFT FeatherWing and (I'm not sure it makes a difference) a Feather Huzzah, flipping the enable switch off and on causes colors to be drawn in BGR mode rather than RGB mode. Flipping it subsequent times has no effect and it is still drawn as BGR. Resetting on either the feather or wing causes it to go back to RGB. I noticed in the bitmapdraw_featherwing example, the touch on the longer axis is mirrored as well.

makermelissa, 2019-04-24:
I noticed this happens with both the ImageReader example and the graphicstest.

ladyada, 2019-04-24:
could be it doenst re-init or half-inits. maybe add a delay at start, to stabilize the display before you start configuring it

makermelissa, 2019-04-24:
Ok, I’ll try that.

makermelissa, 2019-04-25:
Changing the first delay in the init sequence from 10 to 100 seems to have solved the issue.

PR #22 (2019-04-25): Fixed Button Example and issue with reversed colors
Fixes #17 and #21.
closed 2019-04-25 by makermelissa