adafruit/Adafruit_HX8357_Library #18: Graphics test fails on Feather M4 on testRoundRects() test

makermelissa, 2019-04-23:
To reproduce, load the graphicstest_featherwing on a Feather M4 and run. It consistently fails on the testRoundRects() function. Tried using 2 feathers with the same result. This worked fine on a Feather Huzzah.

ladyada, 2019-04-23:
mysterious - can you dig into why?

makermelissa, 2019-04-23:
Yep. I was just digging out some more boards to test on.

makermelissa, 2019-04-23:
This failed on a Feather M0 in the same spot as well.

makermelissa, 2019-04-23:
Oops, looks like I was wrong about which test. It's the testFilledTriangles() that it fails on. Huzzah32 however failed on testRoundRects().

makermelissa, 2019-04-23:
Also, I should note that this is failing on the 1.1.1 version of the driver as well as the sendCommand updates I made. It might be a timing issue.

makermelissa, 2019-04-23:
If it is a timing issue, it may be related to #17 which seemed a little random in its behavior.

makermelissa, 2019-04-23:
Ok, the plot thickens. The issue only occurs when I have an SD card in the TFT FeatherWing.

siddacious, 2019-04-23:
🤔

makermelissa, 2019-04-23:
I’m going to try different SD cards next to see if that makes a difference.

makermelissa, 2019-04-24:
This seems to only happen with a very specific SD card that I happen to have a bunch of. Since it works with the more popular brands and it's not even necessary for this test, I don't think this is worth trying to fix.

closed 2019-04-24 by makermelissa