adafruit/Adafruit_HX8357_Library #21: merged-in touch button example; not working for Feather M4

dsyleixa, 2019-04-24:
re: merged-in touch button example; not working for Feather M4

the M4 board could not be detected because of missing board #defines .

Instead, this works for my Feather M4:
```

/*
 * gfxbuttontest_featherwing
 */

#include <SPI.h>
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_HX8357.h>
#include <Adafruit_STMPE610.h>

#ifdef ESP8266
   #define STMPE_CS 16
   #define TFT_CS   0
   #define TFT_DC   15
   #define SD_CS    2
 
#elif defined ESP32
   #define STMPE_CS 32
   #define TFT_CS   15
   #define TFT_DC   33
   #define SD_CS    14
 
#elif defined TEENSYDUINO
   #define TFT_DC   10
   #define TFT_CS   4
   #define STMPE_CS 3
   #define SD_CS    8
  
#elif defined ARDUINO_STM32_FEATHER
   #define TFT_DC   PB4
   #define TFT_CS   PA15
   #define STMPE_CS PC7
   #define SD_CS    PC5
  
#elif defined ARDUINO_FEATHER52
   #define STMPE_CS 30
   #define TFT_CS   13
   #define TFT_DC   11
   #define SD_CS    27

#elif  defined(ARDUINO_MAX32620FTHR) || defined(ARDUINO_MAX32630FTHR)
   #define TFT_DC   P5_4
   #define TFT_CS   P5_3
   #define STMPE_CS P3_3
   #define SD_CS    P3_2

// Something else!
#elif  defined (__AVR_ATmega32U4__) || defined(ARDUINO_SAMD_FEATHER_M0) || defined (__AVR_ATmega328P__) || defined(ARDUINO_SAMD_ZERO) || defined(__SAMD51__)   
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5

 // optional different defaults 
#else
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5
#endif

#define TFT_RST -1

// Use hardware SPI and the above for CS/DC
[... 314 lines cut ...]
        Serial.println(msg);
         tft.fillRect(DISPLAY_XOFFSET,DISPLAY_YOFFSET, tft.width(), tft.height(), HX8357_BLACK);
        setTextColorIndex(0);
        setTextSizeIndex(-1);
         tftCenterPrint(msg);
        break;
        
      case BTN_RETURN:
        msg = "Menu 1 Ret Button   ";
        Serial.println(msg);
        tftCenterPrint(msg);
        break;
        
      default:
        break;
    }
}

//--------------------------------------------------------------------------------
void loop() {

    processMenu1();
   
}
```

makermelissa, 2019-04-25:
Interesting. The Feather M4 was the first board I tested with and it seemed to be working. Do you want to update your code and submit another PR?

makermelissa, 2019-04-25:
This has been updated.

closed 2019-04-25 by makermelissa

dsyleixa, 2019-04-25:
hello,
all I formerly had found  from you  was
https://github.com/adafruit/Adafruit_HX8357_Library/issues/12#event-2296117523
and
https://github.com/adafruit/Adafruit_HX8357_Library/pull/15
whilst in the Adafruit repo the TFT button example code was already  5 or 7 months old (CMIIW)

Now I tried this version (now 4 hours old):
https://github.com/adafruit/Adafruit_HX8357_Library/blob/master/examples/gfxbuttontest_featherwing/gfxbuttontest_featherwing.ino
and that works fine!

dsyleixa, 2019-04-25:
PS,
in the folder is another file .mega2560.test.skip
https://github.com/adafruit/Adafruit_HX8357_Library/blob/master/examples/gfxbuttontest_featherwing/.mega2560.test.skip
what is that for?

makermelissa, 2019-04-25:
The .mega2560.test.skp file causes the automated test for the Arduino Mega 2560 to be skipped when the code is tested. Since it was designed for a TFT FeatherWing, that was fine.

dsyleixa, 2019-04-25:
but why is that an extra file? 
Why is the code not integrated in the main .ino file?
Finally the code is an example to work for all boards, as long as wired accordingly to either MCU board...?

dsyleixa, 2019-04-26:
PS,
as to .mega2560.test.skp
Why has it such a weird name? (not .ino and 1 leading dot and  1 more dot in between?)
Is that a Arduino code file at all?

makermelissa, 2019-04-26:
The dot at the beginning causes the file to be hidden. The mega2560 means it should only skip the Arduino Mega 2560 test and the test.skp is just the type of file it is. It has no usefulness other than as a way of telling the automated tests which ones to skip. Without it, all tests are run and sometimes we want to skip them.

dsyleixa, 2019-04-26:
Are the automated tests made for end users?
Or can I delete this .mega2560.test.skp file from the lib?
Finally the lib is supposed to be used for end users.

If I delete it and wire accordingly to a Mega pins, will it work for the Mega too, or will it damage the Mega then?

makermelissa, 2019-04-26:
Nope. They’re run whenever we submit changes to reduce the chances of breaking something. You can delete it if it bothers you. It shouldn’t damage the Mega, but it may or may not compile for it. If you’re able to get it compiling then there’s really no reason to have the skip file. It was just included as part of the pull request.

dsyleixa, 2019-04-27:
yes, it compiles for the Mega2560 cpu platform.

makermelissa, 2019-04-27:
Awesome. thanks for letting us know.

dsyleixa, 2019-04-27:
you're welcome, I'm glad if I could help.
Unfortunately some of your contributers or owners (e.g., LadyAda) are acting very unpolite and lock an issue report even if it's not resolved, so follow-up questions or remarks or updates are not possible any more.
That is IMO  - friendly spoken - not helpful at all, neither for your libs nor for the customers purchasing Adafruit products.
Gladfully you are keeping it open until it's reported to be resolved, thank you!

ladyada, 2019-04-27:
@dsyleixa hiya, we cannot do tech support in github issues, if you have technical questions about why a library or hardware is designed a certain way you can ask in the forums, its not for posting in github. bug reports, code contributions only here. thanks!

dsyleixa, 2019-04-27:
well, sometimes for you it seems to be a support request at first glance, but then it turned out to be bug though.
e.g. my report about SPI clock for HX8357:
the freq parameter ti ::begin(freq)  does not represent a frequency at all, at it makes no change at all for either arbitrary value - that's definitely a bug,  but I couldn't report that cause it's locked.

ladyada, 2019-04-27:
bug reports are welcome - if you find a bug, open a specific issue with how to reproduce it (e.g. proof of bug)

dsyleixa, 2019-04-27:
ok, then I will post th issue anew.