adafruit/Adafruit_HX8357_Library #23: very simple touch button example needed

dsyleixa, 2019-04-25:
Still a very simple touch button example needed as the provided source codes are not understandable for a beginner to touch screen as they are overloaded with extra features (e.g., menu arrays).

I tried with that code, but release button is not detected (ts.justReleased() )
after having touched (ts.justPressed() ) and lifting the finger, the button still stays in "pressed" mode, 
one has to touch anywhere else on the screen apart from the button and only after that  the button release is detected.

```
// TFT HX8357:  Adafruit_HX8357.h
// Touchscreen: Adafruit_STMPE610.h

// Programming language: Arduino Sketch C/C++ (IDE 1.8.8)

#include <SPI.h>
#include <Wire.h>
#include <SPI.h>

#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_HX8357.h>
#include <Adafruit_STMPE610.h>

#ifdef ESP8266
   #define STMPE_CS 16
   #define TFT_CS   0
   #define TFT_DC   15
   #define SD_CS    2
 
#elif defined ESP32
   #define STMPE_CS 32
   #define TFT_CS   15
   #define TFT_DC   33
   #define SD_CS    14
   
#elif defined ARDUINO_FEATHER52
   #define STMPE_CS 30
   #define TFT_CS   13
   #define TFT_DC   11
   #define SD_CS    27

// Something else!
#elif  defined (__AVR_ATmega32U4__) || defined(ARDUINO_SAMD_FEATHER_M0) || defined (__AVR_ATmega328P__) || defined(ARDUINO_SAMD_ZERO) || defined(__SAMD51__)   
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5

 // default 
#else
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5
#endif

//-------------------------------------------------------------------------
#define TFT_RST -1

Adafruit_HX8357   tft = Adafruit_HX8357(TFT_CS, TFT_DC, TFT_RST);
Adafruit_STMPE610  ts = Adafruit_STMPE610(STMPE_CS);

//-------------------------------------------------------------------------
// Touch Calibration
/*
#define TS_MINX 3800
#define TS_MAXX 100
#define TS_MINY 100
#define TS_MAXY 3750
*/
#define TS_MINX 100
#define TS_MAXX 3800
#define TS_MINY 3750
#define TS_MAXY 100

 /*
 #define TS_MINX 150
 #define TS_MINY 130
 #define TS_MAXX 3800
 #define TS_MAXY 4000 
*/

//-------------------------------------------------------------------------
Adafruit_GFX_Button button1 /*, button2*/ ;

//-------------------------------------------------------------------------
/*
int16_t  RADMAX=4;
int16_t  offsx=RADMAX, offsy=0;
int16_t  canvasx=400-1, canvasy=320-offsy-1;
*/

#define BLACK       HX8357_BLACK
#define DARKGREY    HX8357_DARKGREY 
#define WHITE       HX8357_WHITE
#define RED         HX8357_RED
#define YELLOW      HX8357_YELLOW
#define CYAN        HX8357_CYAN
#define BLUE        HX8357_BLUE

#define COLOR_BKGR  WHITE // DARKGREY

//-------------------------------------------------------------------------
//-------------------------------------------------------------------------

void setup() {
  int i;

  Wire.begin();
  
  Serial.begin(115200);
  delay(1000);
   
  tft.begin(HX8357D);   
  tft.setRotation(1);
  tft.fillScreen(COLOR_BKGR);
  tft.setTextSize(2);
  
  if (!ts.begin()) {
    Serial.println("Couldn't start touchscreen controller");
    while (1);
  }
  Serial.println("Touchscreen started");   
  
  // create button
  button1.initButton(&tft, tft.width()-40, 20, 70, 40, CYAN, BLUE, YELLOW, "Reset", 2);
  button1.drawButton();  
  
}

//-------------------------------------------------------------------------
//-------------------------------------------------------------------------

void loop() {
  int i;
  float fyaw, fpitch, froll;
  char buf[20]="", buf1[20]="";
     
  TS_Point p;
  p.x=-1; p.y=-1;
  
  // Retrieve a point   
  p = ts.getPoint();
  
  // Scale from ~0->4000 to tft.width using the calibration #'s
  p.x = map(p.x, TS_MINX, TS_MAXX, 0, tft.width());
  p.y = map(p.y, TS_MINY, TS_MAXY, 0, tft.height());
    
    Serial.print("X = "); Serial.print(p.x); Serial.print("\tY = "); Serial.print(p.y);  Serial.print("\tPressure = "); Serial.println(p.z);           
    Serial.println();    
    sprintf(buf, "x=%3d", p.x); sprintf(buf1, "y=%3d", p.y);
    
    tft.setTextColor(RED); 
    tft.fillRect(tft.width()-80,40, 80,40, COLOR_BKGR);  // x1,y1, dx,dy 
    tft.setCursor(tft.width()-75,40);    tft.print(buf);
    tft.setCursor(tft.width()-75,60);    tft.print(buf1);         
  
  button1.press(button1.contains(p.x, p.y)); // tell the button it is pressed
  delay(1);
  
  // ask the buttons if their state has changed 
  if (button1.justPressed()) {
     button1.drawButton(true); // draw invert! 
     tft.setTextColor(RED);    
     tft.fillRect(tft.width()-80,80, 80,40, COLOR_BKGR);  
     tft.setCursor(tft.width()-75, 80);    tft.print("just");
     tft.setCursor(tft.width()-75,100);    tft.print("press");
  }

  if (button1.justReleased() ) {
     tft.fillScreen(COLOR_BKGR);   
     button1.drawButton(); // draw normal 
     tft.setTextColor(RED);    
     tft.fillRect(tft.width()-80,80, 80,40, COLOR_BKGR); 
     tft.setCursor(tft.width()-75, 80);    tft.print("just");
     tft.setCursor(tft.width()-75,100);    tft.print("releas");
    
  }

 
  delay(1);  

}

```

dsyleixa, 2019-05-05:
as an  improvement, one may use kcl1s' changes at the start of loop()
https://forums.adafruit.com/viewtopic.php?f=47&t=151012&p=747057#p747049 :
```
//Mychanges------------------------------------------------
  TS_Point p;
  if (ts.bufferSize())  {
    p = ts.getPoint();
  }
  else  {
    // this is our way of tracking touch 'release'!
    p.x = -1;
    p.y = -1;
    p.z = -1;
  }
  //end my changes-----------------------------------------------
```

final tests and checks are pending but FTM it looks really good!

makermelissa, 2019-05-25:
Since there's already a button example, I'm just going to close this. If anybody wants to share another one, they can submit a PR. :)

closed 2019-05-25 by makermelissa