adafruit/Adafruit_HX8357_Library #24: How to increase SPI clock speed for HX8357?

dsyleixa, 2019-04-26:
How to increase SPI clock speed for HX8357?
The Graphics output is too slow for my purposes.
I am using Arduino IDE 1.8.8 and the Feather M4 and the Feather ESP32 with this Featherwing HX8357 display.

ladyada, 2019-04-26:
https://github.com/adafruit/Adafruit_HX8357_Library/blob/master/Adafruit_HX8357.cpp#L244

closed 2019-04-26 by ladyada

dsyleixa, 2019-04-26:
Sorry, I don't understand:
which faster values for (uint32_t freq) are valid?
which freq are HX8357D or HX8357B?
which is optionally even more faster (e.g. SPI clock 40MHz) ?
Can I pass e.g. 40000000 for 40MHz or 60000000 for 60MHz?

ladyada, 2019-04-26:
you can try any freq you like, technicall the chip has a max of 16000000 i think, but sometimes you can overclock

dsyleixa, 2019-04-26:
if lower than 120MHz, must that be a freq of an integer clock divider (2,3,4,...)  of the M4 cpu clock 
120000000?

ladyada, 2019-04-26:
Arduino picks the closest value yes

dsyleixa, 2019-04-26:
ok,
and now what do HX8357D or HX8357B stand for?

ladyada, 2019-04-26:
hiya we do not do tech support in github issues
if you have tech support questions please post in the forums - this isn't a bug in the library you're asking about!