adafruit/Adafruit_HX8357_Library #25: this current lib 1.1.4 does not compile any more for Feather M4, opposite to the former lib version 1.0.7

dsyleixa, 2019-04-27:
this current lib version does not compile any more for Feather M4 (version=1.1.4)
opposite to the former lib version
version=1.0.7

Arduino: 1.8.8 (Windows 7), 
Board: "Adafruit Feather M4 Express (SAMD51), Enabled"

```
D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=adafruit:samd:adafruit_feather_m4:cache=on -vid-pid=0X239A_0X8022 -ide-version=10808 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_869403 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.openocd.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.CMSIS-Atmel.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -verbose C:\Users\hw\AppData\Local\Temp\arduino_modified_sketch_187767\Ardubench_22G_HX8357_Adafruit.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=adafruit:samd:adafruit_feather_m4:cache=on -vid-pid=0X239A_0X8022 -ide-version=10808 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_869403 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.openocd.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.CMSIS-Atmel.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -verbose C:\Users\hw\AppData\Local\Temp\arduino_modified_sketch_187767\Ardubench_22G_HX8357_Adafruit.ino
Using board 'adafruit_feather_m4' from platform in folder: D:\arduino\portable\packages\adafruit\hardware\samd\1.3.0
Using core 'arduino' from platform in folder: D:\arduino\portable\packages\adafruit\hardware\samd\1.3.0
WARNUNG: Kategorie 'Buttons' in der Bibliothek ButtonClass ist ungültig und wird auf 'Uncategorized' festgelegt
WARNUNG: Kategorie 'TFT menu' in der Bibliothek MenuClass ist ungültig und wird auf 'Uncategorized' festgelegt
WARNUNG: Kategorie '' in der Bibliothek XPT2046 ist ungültig und wird auf 'Uncategorized' festgelegt
Detecting libraries used...
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10808 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.3.0\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.3.0\\variants\\feather_m4" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_869403\\sketch\\Ardubench_22G_HX8357_Adafruit.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10808 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.3.0\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.3.0\\variants\\feather_m4" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.3.0\\libraries\\SPI" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_869403\\sketch\\Ardubench_22G_HX8357_Adafruit.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10808 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.3.0\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.3.0\\variants\\feather_m4" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.3.0\\libraries\\SPI" "-ID:\\arduino\\portable\\sketchbook\\libraries\\Adafruit-GFX-Library-master" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_869403\\sketch\\Ardubench_22G_HX8357_Adafruit.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10808 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-ID:\\arduino\\portable\ [... cut ...]
D:\arduino\portable\sketchbook\libraries\Adafruit_HX8357_Library-master\Adafruit_HX8357.cpp:331:52: error: 'sendCommand' was not declared in this scope

   sendCommand(invert ? HX8357_INVON : HX8357_INVOFF);
[... 214 lines cut ...]

  // Setup the LCD
  tft.begin(HX8357D);
  //tft.begin(2);
  tft.setRotation(3);
  tft.fillScreen(0x0000);
  tft.setTextColor(0xFFFF); tft.setTextSize(1);
  Serial.println("tft started");

  pinMode(tpin1, OUTPUT);
  pinMode(tpin2, OUTPUT);
  pinMode(tpin3, INPUT_PULLUP);

  char  buf[120];
  test(); 
  sprintf (buf, "Ende brickbench");   
  Serial.println( buf);
  TFTprint(buf, 0, 110);
}

void loop() {

}

```

makermelissa, 2019-04-27:
Make sure you have the latest Adafruit_GFX library or it won't be able to find the `sendCommand()` function.

dsyleixa, 2019-04-27:
ok, now it compiles, but the code does not run actually, no screen output at all.

siddacious, 2019-04-27:
@dsyleixa Thank you for reaching out but please use the forums for support. If it is determined that a bug exists and can be reproduced by someone other than you, please open an issue.

closed 2019-04-27 by siddacious

dsyleixa, 2019-04-28:
why do you close this issue report, that is unreasonable whilst it is not resolved!

Actually you should be able to reproduce the issue wth your ownFeatherwing HX8357 and one of your feather M4 boards. It's definitely not a support question, it's a reported bug with a provided test code!

So as long as this issue is not resolved, please re-open it again!

siddacious, 2019-04-28:
@dsyleixa there are many reasons why your code could not display output and finding the exact cause is what the support forums are for.

makermelissa, 2019-04-28:
I agree with @siddacious here. I am not able to reproduce this issue (it compiles and runs fine), so the forums are the appropriate place to address your specific issue.