adafruit/Adafruit_HX8357_Library #26: SPI clock speed for HX8357 can not be changed or increased by Adafruit_HX8357::begin(uint32_t freq),.

dsyleixa, 2019-04-27:
hello,
this is a bug report.

 SPI clock speed for HX8357 can not be changed or increased by Adafruit_HX8357::begin(uint32_t freq),.
(I have the 3.5" HX8357  Featherwing for Feather M4, Arduino IDE 1.8.8)

As to Adafruit_HX8357::begin(uint32_t freq),  
for which freq  do HX8357D or HX8357B stand for, passed as "freq" ?
if I write to Serial it says:
Serial.println(HX8357B);  // --> 11
Serial.println(HX8357D);  //-->  13
But thats no SPI frequency, is it?
So is it a SPI_clock_devider or what else?

anyway,
I can pass any arbitrary value to tft.begin(n), e.g. HX8357B, HX8357D, 40000000, 20000000, or 1 or 2, or anything else;
for a graphic bench mark the output speed stays always the same, either way which value I am passing - so how to make SPI faster?

[code]
// Adafruit Brickbench
// benchmark test for SoCs and MCUs
// PL: GCC,Arduino
// Autor: (C) dsyleixa 2013-2019
//
// change log:
// 2.2.G  excerpt: testing just TFT
// 2.2.   testing both 32fp and 64fp 
// 2.1.1. 32bit fp tests vs. 64bit double (ARM/32bit cores, optional)
//        low-level bitRead/Write vs. digitalRead/Write (AVR cores, optional) 
// 2.1 GPIO r/w
// 2.0 loop counts

#include <SPI.h>
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_HX8357.h>
#include <Adafruit_STMPE610.h>

#ifdef ESP8266
   #define STMPE_CS 16
   #define TFT_CS   0
   #define TFT_DC   15
   #define SD_CS    2
 
#elif defined ESP32
   #define STMPE_CS 32
   #define TFT_CS   15
   #define TFT_DC   33
   #define SD_CS    14
 
#elif defined TEENSYDUINO
   #define TFT_DC   10
   #define TFT_CS   4
   #define STMPE_CS 3
   #define SD_CS    8
  
#elif defined ARDUINO_STM32_FEATHER
   #define TFT_DC   PB4
   #define TFT_CS   PA15
   #define STMPE_CS PC7
   #define SD_CS    PC5
  
#elif defined ARDUINO_FEATHER52
   #define STMPE_CS 30
   #define TFT_CS   13
   #define TFT_DC   11
   #define SD_CS    27

#elif  defined(ARDUINO_MAX32620FTHR) || defined(ARDUINO_MAX32630FTHR)
   #define TFT_DC   P5_4
   #define TFT_CS   P5_3
   #define STMPE_CS P3_3
   #define SD_CS    P3_2

// Something else!
#elif  defined (__AVR_ATmega32U4__) || defined(ARDUINO_SAMD_FEATHER_M0) || defined (__AVR_ATmega328P__) || defined(ARDUINO_SAMD_ZERO) || defined(__SAMD51__)   
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5

 // default 
#else
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5
#endif

#define TFT_RST -1

Adafruit_HX8357 tft = Adafruit_HX8357(TFT_CS, TFT_DC, TFT_RST);
Adafruit_STMPE610 ts = Adafruit_STMPE610(STMPE_CS);

#define  TimerMS() millis()

unsigned long runtime[10];

#define tpin1  11  // GPIO test pins digitalWrite
#define tpin2  12  // GPIO test pins digitalWrite
#define tpin3  13  // GPIO test pins digitalRead

void TFTprint(char sbuf[], int16_t x, int16_t y) {
  tft.setCursor(x, y);
  tft.print(sbuf);
}

int a[500], b[500], c[500], t[500];

    
//--------------------------------------------
inline void displayValues() {

  char buf[120];
  tft.fillScreen(0x0000); // clrscr()

    sprintf (buf, "%3d %9ld  int_Add",    0, runtime[0]); TFTprint(buf, 0, 9);
    sprintf (buf, "%3d %9ld  int_Mult",   1, runtime[1]); TFTprint(buf, 0,18);
    sprintf (buf, "%3d %9ld  fp32_op",    2, runtime[2]); TFTprint(buf, 0,27);
    sprintf (buf, "%3d %9ld  fp64_op",    3, runtime[3]); TFTprint(buf, 0,36);
    sprintf (buf, "%3d %9ld  randomize",  4, runtime[4]); TFTprint(buf, 0,45);
    sprintf (buf, "%3d %9ld  matrx_algb", 5, runtime[5]); TFTprint(buf, 0,54);
    sprintf (buf, "%3d %9ld  arr_sort",   6, runtime[6]); TFTprint(buf, 0,63);
    sprintf (buf, "%3d %9ld  GPIO_togg",  7, runtime[7]); TFTprint(buf, 0,72);
    sprintf (buf, "%3d %9ld  Graphics",   8, runtime[8]); TFTprint(buf, 0,80);
}

//--------------------------------------------
int32_t test_TextOut(){
  int  y;
  char buf[120];
 
  for(y=0;y<10;++y) {   
    tft.fillScreen(0x0000); // clrscr()
    sprintf (buf, "%3d %9d  int_Add",    y, 1000);  TFTprint(buf, 0, 9);
    sprintf (buf, "%3d %9d  int_Mult",   0, 1010);  TFTprint(buf, 0,18);
    sprintf (buf, "%3d %9d  fp32_op",    0, 1032);  TFTprint(buf, 0,27);
    sprintf (buf, "%3d %9d  fp64_op",    0, 1064);  TFTprint(buf, 0,36);
    sprintf (buf, "%3d %9d  randomize",  0, 1040);  TFTprint(buf, 0,45);
    sprintf (buf, "%3d %9d  matrx_algb", 0, 1050);  TFTprint(buf, 0,54);
    sprintf (buf, "%3d %9d  GPIO_togg",  0, 1060);  TFTprint(buf, 0,63);
    sprintf (buf, "%3d %9d  Graphics",   0, 1070);  TFTprint(buf, 0,72);
    sprintf (buf, "%3d %9d  testing...", 0, 1080);  TFTprint(buf, 0,80);
  }
  return y;
}

//--------------------------------------------
int32_t test_graphics(){
  int y;
  char buf[120];
 
 
  for(y=0;y<10;++y) {
    tft.fillScreen(0x0000);
    sprintf (buf, "%3d", y);  TFTprint(buf, 0,80); // outcomment for downwards compatibility

    tft.drawCircle(50, 40, 10, 0xFFFF);
    tft.fillCircle(30, 24, 10, 0xFFFF);
    tft.drawLine(10, 10, 60, 60, 0xFFFF);
    tft.drawLine(50, 20, 90, 70, 0xFFFF);
    tft.drawRect(20, 20, 40, 40, 0xFFFF);
    tft.fillRect(65, 25, 20, 30, 0xFFFF);
    tft.drawCircle(70, 30, 15, 0xFFFF); 

  }
  return y;
}

//--------------------------------------------
long test(){
 unsigned long time0, x, y;
  double s;
  char  buf[120];
  int   i;
  float f;

  

  Serial.println("start test");
  tft.println("start test");
  delay(10);

  // computational benchmark tests 0...7 skipped

  // lcd display text / graphs
  time0=TimerMS();
  s=test_TextOut();  
  s=test_graphics();
  runtime[8]=TimerMS()-time0;
  sprintf (buf, "%3d %9ld  Graphics   ", 8, runtime[8]); 
  Serial.println( buf); 
  tft.println( buf);

  Serial.println();
 
  y = 0;
  for (x = 0; x < 9; ++x) {
      y += runtime[x];
  }
 
  displayValues();
  sprintf (buf, "runtime ges.:  %-9ld ", y);
  Serial.println( buf);   TFTprint(buf, 0,90);
 
  x=50000000.0/y;
  sprintf (buf, "benchmark:     %-9ld ", x);
  Serial.println( buf);   TFTprint(buf, 0,100);

  return 1;
}

//--------------------------------------------
void setup() {
 
  Serial.begin(115200);
  Serial.println("starting Serial()");
  while(!Serial);

  // Setup the LCD
  //tft.begin(HX8357D);
  tft.begin(2);
  tft.setRotation(3);
  tft.fillScreen(0x0000);
  tft.setTextColor(0xFFFF); tft.setTextSize(1);
  Serial.println("tft started");

  pinMode(tpin1, OUTPUT);
  pinMode(tpin2, OUTPUT);
  pinMode(tpin3, INPUT_PULLUP);

  char  buf[120];
  test(); 
  sprintf (buf, "Ende Arduino brickbench");   
  Serial.println( buf);
  TFTprint(buf, 0, 110);
}

void loop() {

}[/code]

makermelissa, 2019-05-19:
Fixed by #30.

dsyleixa, 2019-05-20:
Objection - 
AFAICS, the issue is not fixed yet:
https://github.com/adafruit/Adafruit_HX8357_Library/pull/30 is not about arbitrary clock speed frequencies.
It's just about a constant HX8357B, bat that is just a char value
uint8_t HX8357B = 11

OTOH, 11 is no SPI frequency at all.
Instead, SPI frequencies are e.g. uint32_t 10000000 (10MHz), 30000000 (30MHz), or 120000000 (120MHz).

So how to change the SPI  busfrequency to either 10000000 (10MHz), 30000000 (30MHz), or 120000000 (120MHz)?

dsyleixa, 2019-05-20:
thanks for reopening!

update, tested anew:
as to Adafruit_HX8357::begin

tft.begin(10000000) is appearently slower than ttft.begin() (void) or tft.begin(HX8357D)  or tft.begin(HX8357B),
but .begin(20000000) is as fast as begin(void) or begin(HX8357D)  or begin(HX8357B),
and this can not be made faster than just that, which actually was the intention.

makermelissa, 2019-05-25:
I believe the chip has a maximum frequency of 16MHz. By default, the lib tries to set it to 24MHz and then it falls back to 16MHz. So setting it to 20MHz will have the same result. Attempting to drive it faster will not make it work better, but giving it 240MHz is just way too fast for it. As you have demonstrated, you are able to set it arbitrarily and it responds (just not how you'd prefer) 10MHz is slower than 16MHz, etc. Closing this bug since it's not an issue.

closed 2019-05-25 by makermelissa

dsyleixa, 2019-05-25:
@makermelissa:
thank you for your reply.
Nonetheless, this now sounds different from what you wrote in a prior post, i.e. that SPI can be accelerated to even more than 120MHz though.

ptorrone, 2019-05-25:
@dsyleixa we are committed to providing a friendly, safe and welcoming environment for all.

* be kind and courteous to others
* using welcoming and inclusive language
* being respectful of differing viewpoints and experiences
* collaborating with other community members
* gracefully accepting constructive criticism
* focusing on what is best for the community
* showing empathy towards other community members

your recent comments here and in our forums can be improved. if the antagonistic behavior continues, the next steps will be to ban from our forums and github.

dsyleixa, 2019-05-25:
I don't see your point, tbh, I'm never unfriendly and I am looking for constructive results. I am also dedicated in testing and improving proposals and have already provided several enhancement suggestions.
OTOH, I am a customer of your peoducts, I purchaased lot of items and even am advertising your products widly in our German Robotics forum; nonetheless, if things don't work, a topc has to be issued.