adafruit/Adafruit_HX8357_Library #27: issue https://github.com/adafruit/Adafruit_HX8357_Library/issues/25  is resolved

dsyleixa, 2019-04-30:
issue https://github.com/adafruit/Adafruit_HX8357_Library/issues/25  is resolved,
the example code has the wrong pin definitions.

instead of
#define TFT_CS 10
#define TFT_DC 9
#define TFT_RST 8 // RST can be set to -1 if you tie it to Arduino's reset

would be needed:

#ifdef ESP8266
   #define STMPE_CS 16
   #define TFT_CS   0
   #define TFT_DC   15
   #define SD_CS    2
 
#elif defined ESP32
   #define STMPE_CS 32
   #define TFT_CS   15
   #define TFT_DC   33
   #define SD_CS    14

 
#elif defined TEENSYDUINO
   #define TFT_DC   10
   #define TFT_CS   4
   #define STMPE_CS 3
   #define SD_CS    8
  
#elif defined ARDUINO_STM32_FEATHER
   #define TFT_DC   PB4
   #define TFT_CS   PA15
   #define STMPE_CS PC7
   #define SD_CS    PC5
  
#elif defined ARDUINO_FEATHER52
   #define STMPE_CS 30
   #define TFT_CS   13
   #define TFT_DC   11
   #define SD_CS    27

#elif  defined(ARDUINO_MAX32620FTHR) || defined(ARDUINO_MAX32630FTHR)
   #define TFT_DC   P5_4
   #define TFT_CS   P5_3
   #define STMPE_CS P3_3
   #define SD_CS    P3_2

// Something else!
#elif  defined (__AVR_ATmega32U4__) || defined(ARDUINO_SAMD_FEATHER_M0) || defined (__AVR_ATmega328P__) || defined(ARDUINO_SAMD_ZERO) || defined(__SAMD51__)   
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5

 // default 
#else
   #define STMPE_CS 6
   #define TFT_CS   9
   #define TFT_DC   10
   #define SD_CS    5
#endif

#define TFT_RST -1

ladyada, 2019-04-30:
thanks, please submit a PR for the fix!

dsyleixa, 2019-04-30:
no idea how to do that - what is a PR?

ladyada, 2019-04-30:
all good, we have 2 excelleng guides, and this is a great (and key) skill to pick up
https://learn.adafruit.com/an-introduction-to-collaborating-with-version-control
https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

dsyleixa, 2019-04-30:
hmm - too complicated, an my English skills are far too poor.
But now you eventually have my bug fix so you might fix that easilly for your next release!

ladyada, 2019-04-30:
try it! our guides are very detailed but the process isnt that complex

makermelissa, 2019-04-30:
In which example file are the pins defined incorrectly?

dsyleixa, 2019-04-30:
no, thanks!
I reported he bug and the fix, 
I fixed that for myself, so all right now to me!
Feel free to copy that for your own purposes!

dsyleixa, 2019-04-30:
it was  in the graphicstest

dsyleixa, 2019-04-30:
PS,
but also graphicstest_featherwing is missing the SAMD21 #defines

makermelissa, 2019-04-30:
The graphicstest example is for the breakout and is intended to have pin numbers changed if it suits you. graphicstest_featherwing is meant if you have the FeatherWing, which it sounds like you do. I will try a feather M0 on it a little later.

makermelissa, 2019-05-01:
It works fine on both the Feather M4 using `defined(__SAMD51__)` and Feather M0 using `defined(ARDUINO_SAMD_ZERO)`, but was missing a check for the nRF52840 Express, which I have added as `defined(ARDUINO_NRF52_FEATHER)` and will submit a PR.

closed 2019-05-01 by makermelissa

dsyleixa, 2019-05-01:
yes, indeed mixing all breakout and featherwing examples in 1 single examples folder is confusing for users,
perhaps I would recommend to create to 2 diffent subfolders called
"HX8357breakout_examples" and
"HX8357featherwing_examples"
containing just the proprietary subset of example souces.

BTW,
`#ifdef SAMD_SERIES  `
covers all SAMD boards (M0==SAMD21, M4==SAMD51)

whilst
`#ifdef SAM `
covers all ARM Cortex SAM boards, e.g. SAMD21, SAMD51, ASAM3X8E (Due)

perhaps one or the other might make your examples more readable.... 8)

makermelissa, 2019-05-02:
Good ideas. :)