adafruit/Adafruit_HX8357_Library #38: Font color background no longer acceptable work around

DJDevon3, 2020-07-09:
- Arduino board:  NRF52840 Feather Sense
- Arduino IDE version: 1.8.13

Font color work around using solid font background for refreshing dynamic character data only works on solid color background.
`    tft.setTextColor(HX8357_RED, HX8357_BLACK);
    tft.setCursor(0, 40);
    tft.setTextSize(3);
    tft.println("Temp: ");`

As you likely know without the background color present the character data will just overwrite itself until you're left with a solid filled in block. However when specifying a background color the character clears the background and redraws itself fine.

Now that TFT's are getting much larger, processors are getting faster, we can add wallpaper backgrounds. The way fonts were displayed worked well in the past because people used solid color backgrounds on the entire TFT. Now that wallpapers can be used we really need a font update that doesn't require a background color to display correctly.   

[image]

ladyada, 2020-07-09:
youcould use circuitpython displayio OR littlevgl, this library does not have dynamic background redrawing capability

closed 2020-07-09 by ladyada

PaintYourDragon, 2020-07-10:
Use GFXcanvas16 to create offscreen images, render into that, copy to display with drawRGBBitmap(). This is the Official And Canonical Way of handling it in GFX lib, which is simply a passive framebuffer and doesn’t abstractify graphical elements (like the aforementioned DisplayIO or LittlevGL, the latter produces _really nice_ antialiased text, check it out). Relying on background color to overwrite text is a Fundamentally Bad Idea in a world of proportionally-spaced fonts and I wish people would avoid using that as a crutch, it’s not a VIC-20. The background color functionality only remains in the library to maintain compatibility with older code.